cbalint13/tinnymodbus #2: Some questions about your project

ZigFisher, 2018-12-11:
Dear Sir !

Your project is simple and beautiful, my congratulations !
I want to ask if you will have support from other I2C sensors ?
I have a small telemetry project - https://github.com/ZigFisher/Glutinium/tree/master/i2c-telemetry/src . 
I saw your implementation and I really liked it.
I would like to see in her INA219, ADS1115, PCF8574, AM2321, BMP085, DS1621, LM75.....

TNX !

cbalint13, 2018-12-23:
@ZigFisher ,

Thanks for your interest in this project !

   Will continue to add features, need to publish on main page a sort of timeline for all goals.

* I try lineup few of my related thoughts:

1) Add driver that can do GPIO (in/out control or pulse counter) and A/D (analog values).
2) Add more I2C drivers (SPI if can do it on 2 wire only e.g. miso+mosi can be mixed).
3) Make drivers modular, allow user to choose and compile custom list of them until it fits the memory.
4) Expose a simple API for ```devs/``` so anyone can contribute drivers using simple rules.

On my very immediate list would be gpio+adc (1) & some rfid readers support, time when drivers should become modular (otherwise outfit 8k memory).

In addition:

* Document more for people having trouble compiling and using this project (toolchain requires gcc8).

* Will see how to make commercial samples of these small boards for those who desire, and document some interesting real case scenarios where it was used so far.

* No plans to write external C tools, would exceed the goal of the project. I believe python is cross-platform enough for module communication, perhaps I would write a python variant too for the existing firmware uploader.

stif, 2019-09-04:
Hi Cristian,

> Will see how to make commercial samples of these small boards for those who desire, and document some interesting real case scenarios where it was used so far.

I am interested in some samples, maybe you have some for sale?

Kind Regards,
Stefan

cbalint13, 2019-09-04:
> Hi Cristian,
> 
> > Will see how to make commercial samples of these small boards for those who desire, and document some interesting real case scenarios where it was used so far.

I intend the same.

> 
> I am interested in some samples, maybe you have some for sale?

Will be available soon on various platform including Tindie.

> 
> Kind Regards,
> Stefan

Daniel091988, 2022-02-28:
Hello, 
I am not a programmer I am an user...
I need to know how can I use this module. I need the i2c communication and the 1 wire communication, but can I use additional a Digital input/output and or an Analog Input/output?

I want to build an PIR sensor for my planned smarthome. 
it needs an input for the PIR sensor, an i2c input for an temperature and humidity sensor and an input for a brightness sensor.
If possable an input for an audio sensor or any other input.

Thank you

closed 2024-02-27 by cbalint13