cbalint13/tinnymodbus #6: It is not working stable.

alvor, 2019-11-13:
Hi! It is not working stable because built-in oscillator cannot provide the required frequency accuracy. Only one of the four chips worked. Quartz resonator needed. But it will take two pins. At least Attiny84 needed.

cbalint13, 2019-11-13:
At so low speed the jitter is ignorable. I dont have any problems reading from dozens (almoust 100 now) of such modules since 2 year. Continuous aquisition per minute basis.
I agree that during the flash procedure (higher speed, pure soft IRQ) skews may occur 1:20 rate, but flash can be restarted.

alvor, 2019-11-21:
Very strange ... I tried to get the bootmode-info.py script to work properly. I have only one out of four working fine at a speed of 9600. Another one works every other time. Two more do not work at all. Maybe I'm doing something wrong?

cbalint13, 2019-11-21:
@alvor,

What sensors are you reading ?

* Please make **sure** you put the decoupling capacitor at the output of voltage supply (5V & GND).
* Also make sure you have at least 6V on the input of the module, if you modbus line is very long even the starting 12V may drop down.

The mentioned capacitor (C2) in the schematic is trivial, otherwise you will experience missreadings and timeouts on reading due to noise induced by sensor's power rail.

alvor, 2019-11-21:
Sensors did not even try to read. Try only bootmode-info.py . I made my board. But the scheme is the same.

alvor, 2019-11-21:
[image]

cbalint13, 2019-11-22:
@alvor,

* Would you try please to concatenate boot.hex + main.hex and flash resulting .hex into attiny ?
* Make sure what you flash in attiny is correct, eventrualy dump whole out and see boot+main is there.
* Also it is important to flash the eeprom part boot.eep file which holds the 0x01 value (Modbus ID).

Then try read sensors directly.

I can send you 3 samples of mine (just write me in PM) in the worst case.

Pomelo81, 2020-03-14:
how can i Change the ds18....20 Modbus Register or the ID of 1Wire devices? 

the bus becames a error if iobroker  reading 0x200 2 regs.  ; 0x201 2 regs…
now i want to Change the id or Register to +1.
id 00  reg 0x200
id 02 reg  0x202 ……
or
id 00 reg 0x200
id 01 reg 0x202 …..

i m not the bestz programmer…. can you help me for my Problem???
thanks

storfusker, 2020-10-27:
Maybe alvor is right when he claims that there can be some unstability with the internal oscillator
From what I can read, then the internal oscillator is factory calibrated at 3V, and if 5V is used for the power, then the oscillator slows down as much as 10-15%
This can calibrated by purring the oscillator signal out on a pin, and calibrated by a simple sw command either static in the code, or a byte saved in EEPROM and read at startup
Br
Steen

closed 2024-02-27 by cbalint13