cbalint13/tinnymodbus #11: Can't read sensors

Fux2, 2021-07-23:
Hello,
I build up two of the tinnymodbus and flashing over modbus works fine. I tested a DS18S20 and a I2C SHT21 sensor. Both can't be read. When using the Python script "" the Attiny hangs up afterwards and power has to be removed and connected again for further working. I tried also an older firmware - same results. How can I debug the readout of the sensors or do you have any suggestions? Any help is appreciated.
$ python3.8 ./sensors-i2c-read.py 
0x03 0x0000

0  running mode

0x03 0x0001

0.01  software version 

0x03 0x0002

1  slave address

0x03 0x0003

4.989590167999268  V (internal)

0x03 0x0004

39.000003814697266  C (internal)

0x04 0x1200

-46.849998474121094  C (sht21)

0x04 0x1201

-6.0  % (sht21)

0x04 0x1210

0.0  lux VI (si1145)

0x04 0x1211

0.0  lux IR (si1145)

0x04 0x1212

0.00 UV index (si1145)

0x04 0x1220

 0.000000 VI lux (bh1750)

0x04 0x1230

 0.00 C (bmp280)

0x04 0x1231

 0.00 hPa (bmp280)

0x04 0x1240

No BME280 found.
`
`

cbalint13, 2021-08-19:
@Fux2 ,

Most common mistakes:

* Check power supply make sure you have ~5V right on the Vdd pin of Attiny75 (diode & cable-length drops down significantly).
* Check that C1, C2 caps are present on the PCB  on booth input and output pins, filtering out parasitic is _crucial_ .

The only way to further debug is to attach a signal analyser, since signals are slow a low end debugger is fine.
I can recommend to look into [sigrok](https://sigrok.org/wiki/Supported_hardware) backend & [pulseview](https://sigrok.org/wiki/PulseView) GUI supported ones.

no-ahnu, 2024-01-13:
Unfortunately I cannot read any data from the BME280.
The following is output on the BMP280:
0x04 0x1230
  0.00 C (bmp280)
0x04 0x1231
  0.00 hPa (bmp280)

The BME280 says “No BME280 found”.

To test I changed the address of BMP280 to 76 ( [here](https://github.com/cbalint13/tinnymodbus/blob/fdc02d94a7130ce141afaea2893fd8141a09635f/devs/bmp280.h#L58C1-L58C28) ). My Raspberry displayed this address on I2C.

Now the measured values from the BMP280 are correct, but the BME280 still says "No BME280 found".

0x04 0x1230
  21.29C (bmp280)
0x04 0x1231
  1012.02 hPa (bmp280)

0x04 0x1240
No BME280 found.

Do you have a tip for me? Thanks

cbalint13, 2024-01-13:
@no-ahnu ,

Did you enable ```-DBME280``` flag (and may remove -DBMP280) in the make.sh  script ?
https://github.com/cbalint13/tinnymodbus/blob/024dad54e0ae67c3fa570620e4a8934f21528c19/make.sh#L21

no-ahnu, 2024-01-13:
Hello cbalint13 and thank you for the quick answer. I think the project is very great.

I changed the line in Make.sh. `DEVS_ENABLE="-DDS18B20 -DBME280"`

0x04 0x1230
No BMP280 found.

0x04 0x1240
  11018751.18C (bme280)

0x04 0x1241
  11490358.07 hPa (bme280)

0x04 0x1242
  11118287.26 %RH (bme280)

The BME280 now delivers values, but unfortunately the wrong ones.

cbalint13, 2024-01-13:
> Hello cbalint13 and thank you for the quick answer. I think the project is very great.
> 
> I changed the line in Make.sh. `DEVS_ENABLE="-DDS18B20 -DBME280"`

* May also try: `DEVS_ENABLE="-DDS18B20 -DBME280 -DCALIBRATION"`
* Where this -DCALIBRATION adds some extra options to calibrate the values.

> 
> 0x04 0x1230 No BMP280 found.
> 
> 0x04 0x1240 11018751.18C (bme280)
> 
> 0x04 0x1241 11490358.07 hPa (bme280)
> 
> 0x04 0x1242 11118287.26 %RH (bme280)
> 
> The BME280 now delivers values, but unfortunately the wrong ones.

I don't own/use ```bme280``` variant , since it is untested by me i left this as a experimental driver.

@stif , do you have ideas what might be wrong with these bad values ?

no-ahnu, 2024-01-13:
thanks for the tip.
I activated the "-DCALIBRATION", unfortunately no success:

user@rtu:~/tinnymodbus/tools$ sudo python3 examples/pymodbus3.x/sensors-read2.py
0x03 0x0000
0 running mode

0x03 0x0001
0.06 software version

0x03 0x0002
1 slave address

0x03 0x0003
5.166972637176514V (internal)

0x03 0x0004
38.7674446105957 C (internal)

1-wire
0x04 0x0001
2 devices found
   id: 0 0x04 0x0100
   28b0315704f53ceb
   id: 0 0x04 0x0200
   22.75 Celsius

   id: 1 0x04 0x0101
   289e3581e37c3c79
   id: 1 0x04 0x0201
   22.5 Celsius

i2c
0x04 0x1240
  11018698.75C (bme280)
0x04 0x1241
  11490341.68 hPa (bme280)
0x04 0x1242
  11118287.26 %RH (bme280)

I hope for an answer from stif

cbalint13, 2024-01-13:
> thanks for the tip. I activated the "-DCALIBRATION", unfortunately no success:

> i2c 0x04 0x1240 11018698.75C (bme280) 0x04 0x1241 11490341.68 hPa (bme280) 0x04 0x1242 11118287.26 %RH (bme280)
 
> I hope for an answer from stif

* Could you try this patch / changes below (with ```DEVS_ENABLE="-DDS18B20 -DBME280"```)  ?
```
diff --git a/main.c b/main.c
index 78a72df..db92e85 100644
--- a/main.c
+++ b/main.c
@@ -681,21 +681,21 @@ int main(void)
                                       bme280_done = 0x01;
                                     }
 
-                                    float V;
+                                    int32_t V;
                                     if ( daddr == 0x1240 )
                                     {
-                                      V = (float) bme280_read_value( BME280_TEMP )/100;
+                                      V = bme280_read_value( BME280_TEMP );
                                       #ifdef CALIBRATION
                                       V =  V + CAL_TOffset;
                                       #endif
                                     }
                                     if ( daddr == 0x1241 )
                                     {
-                                      V = (float) bme280_read_value( BME280_PRES )/100; 
+                                      V = bme280_read_value( BME280_PRES );
                                     }
                                     if ( daddr == 0x1242 )
                                     {
-                                      V = (float) bme280_read_value( BME280_HUM )/100;
+                                      V = bme280_read_value( BME280_HUM );
                                       #ifdef CALIBRATION
                                       V =  V + CAL_HOffset;
                                       #endif
```
* I suspect the issue is the cast of ```float```, BMP280 use ```int32_t``` (avoiding expensive floats in the firmware)

no-ahnu, 2024-01-13:
Here I first have to understand what I should do. But I will definitely try it. Thanks for the tip!

cbalint13, 2024-01-13:
> Here I first have to understand what I should do. But I will definitely try it. Thanks for the tip!

@no-ahnu ,

I just pushed this commit upstream, you can update your local copy now with ```git pull``` & recompile it.

Some notes:
* This change assure integer only math (as the driver was designed) and avoid very expensive ```float``` in the drivers.
* This is now consistent with the fact that on remote side a ```int32``` type is expected on the modbus wire, see:
* https://github.com/cbalint13/tinnymodbus/blob/master/tools/examples/pymodbus3.x/sensors-i2c-read.py#L211

Let me know about the outcome of results on your side.

no-ahnu, 2024-01-13:
I just saw that you have already made the changes to main.c. Then I no longer have to think about what needs to be adjusted.
I have now downloaded the new main.c and created a new main.hex with `DEVS_ENABLE="-DDS18B20 -DBME280"` in make.sh.

Flashed this and here is the output.

user@rtu:~/tinnymodbus/tools$ sudo python3 examples/pymodbus3.x/sensors-read2.py
0x03 0x0000
0 running mode

0x03 0x0001
0.06 software version

0x03 0x0002
1 slave address

0x03 0x0003
5.119999885559082 V (internal)

0x03 0x0004
38.7674446105957 C (internal)

1-wire
0x04 0x0001
2 devices found
   id: 0 0x04 0x0100
   28b0315704f53ceb
   id: 0 0x04 0x0200
   22.9375 Celsius

   id: 1 0x04 0x0101
   289e3581e37c3c79
   id: 1 0x04 0x0201
   22.5625 Celsius

i2c
0x04 0x1240
  21.81C (bme280)

0x04 0x1241
  1011.56 hPa (bme280)

0x04 0x1242
  49.29%RH (bme280)

That looks really GREAT. THANKS!!!

cbalint13, 2024-01-13:
@no-ahnu 
 
> i2c 0x04 0x1240 21.81C (bme280)
> 
> 0x04 0x1241 1011.56 hPa (bme280)
> 
> 0x04 0x1242 49.29%RH (bme280)
> 
> That looks really GREAT. THANKS!!!

Thank you too for the time and patience !

closed 2024-01-13 by cbalint13