cbalint13/tinnymodbus #17: Digispark ATTINY85 Mini USB

saorxxx1, 2022-10-11:
Hello Balint Cristian,

First of all, congratulations for this work. Nice work. Thank you for your effort.

My problem is how do I load this work on an ATTINY85 module. There is a USBasp programming kit, I tried it on linux. The error message related to the make command is "overlap boot begin". I guess I didn't succeed. Are there any installation instructions on this?

Thank you for your interest and I wish you good work.
Kind regards.

DrRob, 2022-10-12:
I think this "overlap" issue happens if you don't use the recommended version of the toolchain - the code is a very tight fit and others versions of the toolchain produce an image that is too large to fit, causing it to overlap the area reserved for the bootloader.

saorxxx1, 2022-10-12:
> Araç zincirinin önerilen sürümünü kullanmazsanız bu "örtüşme" sorununun ortaya çıktığını düşünüyorum - kod çok sıkı bir şekilde oturuyor ve araç zincirinin diğer sürümleri sığamayacak kadar büyük bir görüntü üretiyor ve bu da alanın üst üste gelmesine neden oluyor önyükleyici için ayrılmıştır.

Hello DrRob,

First of all, thank you for your interest and response. It's clear that I really need some help with this. I don't understand exactly what you mean, but there seems to be an incompatibility as you say. I loaded the existing boot.hex and boot.eep files with USBasp. But I could not solve this problem with the make command and the burn command.

DrRob, 2022-10-13:
Hi Saorxxx1,

Have a look at this closed issue - I think it's the same problem: https://github.com/cbalint13/tinnymodbus/issues/10

cbalint13, 2022-10-25:
Hi @saorxxx1,

  * You can use USBasp programmer with averdude, this is the way how I do it.
  * Make sure you compile the project with gcc 8.0.1 , otherwise will have issue with size.
  * It is unfortunate that newer GCC > 8.0.1 optimize badly despite -Os the size for this project.
 
If not sure how to obtain exact 8.0.1 you can use ready compiled .hex provided here.

closed 2022-10-25 by cbalint13

no-ahnu, 2024-01-09:
Hello,
i use the Google translator.
Unfortunately I fail when trying to run make.sh. I also get the "error" overlap. Unfortunately I can't get gcc version 8.0.1 installed either. With versions 7.4, 8.4 or 9 I get the error message.

What else can I do? or can someone send me the boot.hex and main.hex?

THANK YOU and greetings

cbalint13, 2024-01-09:
> Hello, i use the Google translator. Unfortunately I fail when trying to run make.sh. I also get the "error" overlap. Unfortunately I can't get gcc version 8.0.1 installed either. With versions 7.4, [...]

Hi @no-ahnu ,

You can follow these steps for a very easy dockerized deployment:
https://github.com/cbalint13/tinnymodbus/tree/master/tools/docker

Now,
* You can choose avr-gcc-8 for a safe default way.
* If you wish avr-gcc-latest or smaller binary you can edit the drivers here:
https://github.com/cbalint13/tinnymodbus/blob/master/make.sh#L21

I've also updated the main page to point to these build steps.

Greetings,
~Cristian.

no-ahnu, 2024-01-09:
THANKS Cristian for the quick response and help!

I was able to create the file using Docker :). 
Unfortunately I cannot fully execute the remote flash. It lingers

3 or 5 TX: RV: Lines stand. Do you also have a tip here?
Or can the main.hex also be flashed via the programmer?

no-ahnu, 2024-01-13:
I solved the problem with a new Modbus <-> USB converter with CH343 chip

cbalint13, 2024-01-13:
> I solved the problem with a new Modbus <-> USB converter with CH343 chip

To fully answer, I used many kind of ```usb <-> rs485``` or ```rs232 <-> rs485```, from cheapest max487 like the one here to expensive galvanic isolated solutions (ADUM series) and never had problems. As a practice when in doubt is easy to double check things with a scope (if have one at hand), these are realy slow / lazy signals easy to check md verify.