cbalint13/tinnymodbus #19: Program Pointer an EEPROM Question

stif, 2022-11-03:
Hi Christian,

i try to learn from and adapt your excellent code. i was adding a humidity offset value into the eeprom, which is working only partly.
It is working inside main program, but the mainmode<->bootmode switch is not working afterwards, and i hope you can explain a little bit how it works.

For Example this part [Line 615 in main.c](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/main.c#L615):
```
// jump into bootloader (0x1C00)
"  ldi   r30, 0x00                      \n\t"
"  ldi   r31, 0x0E                      \n\t"
"  ijmp                                 \n\t"
```
i guess it load `0x0E00` into Z register and then jump to this memory address to continue the program. But why you dont jump to `0x1C00`? What i am missing here?

Also in the opposite direction i dont get why you not jump to `0x0000` but instead to `0x000F` [Line138 in boot.c](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/boot.c#L138)

And also which part i have to change if i want to add another byte to the eeprom? Do i have to alter the make.sh script [somewhere here](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/make.sh#L50) and how?

Would be great if you find some time to clarify my questions a bit!

Thank you,
Stefan

cbalint13, 2022-11-03:
Hi Stefan,

> i try to learn from and adapt your excellent code. i was adding a humidity offset value into the eeprom, which is working only partly. It is working inside main program, but the mainmode<->bootmode [...]

* Before switching from ```main``` to ```boot```  there is a mark (string) pushed into the stack [here](https://github.com/cbalint13/tinnymodbus/blob/master/main.c#L605-L613)
* Now ```boot``` [check for this marker](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/boot.c#L121-L134) and if it is present **stays** in ```boot``` otherwise will jump back to ```main```.

The marker in stack is used to track origin of jump, if it was a real request originating from ```main```.
Now  using ```ijmp``` (with Z preloaded) make sure stack marker is not altered, I found this way of doing it.
You can get rid of this marker since attiny always coldboot with ```main``` and not ```boot```, I added it for future purposes.

> 
> Also in the opposite direction i dont get why you not jump to `0x0000` but instead to `0x000F` [Line138 in boot.c](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/boot.c#L138)

* On 0x00000 <-> 0x0000F there are the start+interrupt vectors
* So right after, at 0x000F is the begining of ```main``` zone contents
* The ```boot``` is installed at 0x1C00 near the very end of flash memory

To see ```main```:
```
$ avr-objdump -S -D build/main.elf
00000000 <__vectors>:
       0:       10 c0           rjmp    .+32            ; 0x22 <__ctors_end>
       2:       2a c0           rjmp    .+84            ; 0x58 <__bad_interrupt>
       4:       75 c2           rjmp    .+1258          ; 0x4f0 <__vector_2>
       6:       28 c0           rjmp    .+80            ; 0x58 <__bad_interrupt>
       8:       27 c0           rjmp    .+78            ; 0x58 <__bad_interrupt>
       a:       09 c3           rjmp    .+1554          ; 0x61e <__vector_5>
       c:       25 c0           rjmp    .+74            ; 0x58 <__bad_interrupt>
       e:       24 c0           rjmp    .+72            ; 0x58 <__bad_interrupt>
      10:       23 c0           rjmp    .+70            ; 0x58 <__bad_interrupt>
      12:       22 c0           rjmp    .+68            ; 0x58 <__bad_interrupt>
      14:       21 c0           rjmp    .+66            ; 0x58 <__bad_interrupt>
      16:       20 c0           rjmp    .+64            ; 0x58 <__bad_interrupt>
      18:       1f c0           rjmp    .+62            ; 0x58 <__bad_interrupt>
      1a:       1e c0           rjmp    .+60            ; 0x58 <__bad_interrupt>
      1c:       8b c2           rjmp    .+1302          ; 0x534 <__vector_14>

0000001e <SWVers>:
      1e:       30 2e           mov     r3, r16
      20:       30 33           cpi     r19, 0x30       ; 48

00000022 <__ctors_end>:
      22:       11 24           eor     r1, r1
      24:       1f be           out     0x3f, r1        ; 63
      26:       cf e5           ldi     r28, 0x5F       ; 95
      28:       d2 e0           ldi     r29, 0x02       ; 2
      2a:       de bf           out     0x3e, r29       ; 62
      2c:       cd bf           out     0x3d, r28       ; 61

0000002e <__do_copy_data>:
      2e:       10 e0           ldi     r17, 0x00       ; 0
```

* Notice that 0x000F(as 16bit program counter) meaning 0x001e (octet position) is after all vectors ends.
* The entry into ```main``` is @ <__ctors_end> as a ```rjmp```  installed at very 0x00000 which points to that.

Not sure why not used 0x0000, **the legit way** (as you noticed), but it seems that 0x000F also works (a bug ?!).
Going to 0x000F  via <SWVers> and then <__ctors_end> works until compiler moves <__ctors_end> badly away :-)

> And also which part i have to change if i want to add another byte to the eeprom? 

* Just declare more variables (or even structures if like it better) tagged ```EEMEM``` [here](https://github.com/cbalint13/tinnymodbus/blob/master/libs/eeprom.c#L53) and [header here](https://github.com/cbalint13/tinnymodbus/blob/master/libs/eeprom.h#L54).

* Make sure you **always** use dedicated ```eeprom_read_*()``` dedicated functions to [access the eeprom data](https://github.com/cbalint13/tinnymodbus/search?q=Idmodbus).

* In case of structure you can force cast it back to it's structure template after eeprom_read_block().

Functions here: https://www.nongnu.org/avr-libc/user-manual/group__avr__eeprom.html
Good walkthrough here: https://www.avrfreaks.net/sites/default/files/forum_attachments/EEPROM.pdf

> Do I have to alter the make.sh script [somewhere here](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/make.sh#L50) and how?
No, that line only dumps/copy out eeprom content to a separate boot.eep file (for easy viewing of content).

> 
> Would be great if you find some time to clarify my questions a bit!

Hope to make sense for you, please feel free to ask more questions if needed.

> 
> Thank you, Stefan

Cheers,
~Cristian

closed 2022-11-04 by cbalint13

stif, 2022-11-08:
Hello Cristian,

thank you very much for your detailed answer!

> First, 0x0E00 from AVR's PC (program counter) 16bit view means 0x1C00 in 8bit octet view !

I still did not get why (16bit view to 8bit octet view makes no sense for me), but good to know for whatever reason i have to divide the HEX Number from `avr-objdump` by 2 to get the HEX number i need to address with the asm code.. (eg. to jump to address `0x001E` from `avr-objdump` i have to divide the address by 2 and use the resulting `0x000F` in the asm code for the jump)

Also thank you for the `avr-objdump` command, now i know how where to look at when i need the addresses for the different code parts.

>Not sure why not used 0x0000, the legit way (as you noticed), but it seems that 0x000F also works (a bug ?!).
Going to 0x000F via SWVers and then __ctors_end works until compiler moves <__ctors_end> badly away :-)

I tried it. when i alter boot.c to jump to `0x0000` instead of `0x000F` the main.c program will not run, so for some reason it must jump to `0x000F` in order to work correctly. (Funny fact: if i jump from boot.c to `0x0011` aka `<__ctors_end>` directly, it will work. so it seems the `0x0000` jump to `<__ctors_end>` will not work for some reason..)

Also adding a variable to the eeprom.[c|h] is leading to a non working code. See the diff here: https://github.com/stif/tinnymodbus/compare/stif:master..stif:Humidity-Offset
The Problem is: When the eeprom is 2 bytes in size there is some overlapping in the compiled code:
```
Disassembly of section .eeprom:

00810000 <HumidityOffsetEE>:
	...

00810001 <IDModbus>:
  810001:	Address 0x0000000000810001 is out of bounds.

```
Unfortunately i dont know where to adjust the bonds and position of the eeprom, but i guess [somewhere here](https://github.com/cbalint13/tinnymodbus/blob/5b1e3b71b35a69f4989b76e1052f2cca09b05cd8/make.sh#L50):
```
avr-objcopy -j .eeprom --no-change-warnings --change-section-lma .eeprom=0x0000 -O ihex build/main.elf boot.eep
```

Cheers,
Stefan

cbalint13, 2022-11-08:
> Hello Cristian,
>
> thank you very much for your detailed answer!
>
>
> I still did not get why (16bit view to 8bit octet view makes no sense for me), but good to know for whatever reason i have to divide [...]

Instructions of AVR ISA are 16 bit long, so PC pointer is also 16bit long, CPU always steps one instruction over 16bit (2 octets).

> 
> Also thank you for the `avr-objdump` command, now i know how where to look at when i need the addresses for the different code parts.
> 
> > Not sure why not used 0x0000, the legit way (as you noticed), but it seems that 0x000F also works (a bug ?!).
> > Going to 0x000F via SWVers and then __ctors_end works until compiler moves <__ctors_end> badly away :-)
> 
> I tried it. when i alter boot.c to jump to `0x0000` instead of `0x000F` the main.c program will not run, so for some reason it must jump to `0x000F` in order to work correctly. (Fun fact: if i jump from boot.c to `0x0011` aka `<__ctors_end>` directly, it will work. so it seems the `0x0000` jump to `<__ctors_end>` will not work for some reason..)

 That sounds weird, it should work.

 * The view is on the locally produced elf/hex file.
 * Maybe vector @ 0x00000 is not installed after flash ? 
 * I remember [forcing the omission for vector table](https://github.com/cbalint13/tinnymodbus/blob/master/crt1.S#L60-L68) in case of ```boot``` image, but not the ```main```.

   One way is to debug on AVR, I don't have that possibility.
   We could also check by dumping AVR's content back (avrdude), and re-dissasamble (like below):

```
$ avr-objdump -D -S -m avr5 main.hex | more

main.hex:     file format ihex

Disassembly of section .sec1:

00000000 <.sec1>:
       0:	10 c0       	rjmp	.+32     	;  0x22
       2:	2a c0       	rjmp	.+84     	;  0x58
       4:	2b c3       	rjmp	.+1622   	;  0x65c
       6:	28 c0       	rjmp	.+80     	;  0x58
       8:	27 c0       	rjmp	.+78     	;  0x58
       a:	bf c3       	rjmp	.+1918   	;  0x78a
       c:	25 c0       	rjmp	.+74     	;  0x58
       e:	24 c0       	rjmp	.+72     	;  0x58
      10:	23 c0       	rjmp	.+70     	;  0x58
      12:	22 c0       	rjmp	.+68     	;  0x58
      14:	21 c0       	rjmp	.+66     	;  0x58
      16:	20 c0       	rjmp	.+64     	;  0x58
      18:	1f c0       	rjmp	.+62     	;  0x58
      1a:	1e c0       	rjmp	.+60     	;  0x58
      1c:	41 c3       	rjmp	.+1666   	;  0x6a0
      1e:	30 2e       	mov	r3, r16
```

> 
> Also adding a variable to the eeprom.[c|h] is leading to a non working code. See the diff here: https://github.com/stif/tinnymodbus/compare/stif:master..stif:Humidity-Offset The Problem is: When the eeprom is 2 bytes in size there is some overlapping in the compiled code:

EEMEM tag will always map to the beginning of eeprom section thus all declared vars will be there (yes overlapping !).
To avoid this overlap you either map a single array or a single bigger struct, some good references:
 * struct: https://www.avrfreaks.net/forum/eemem-and-structs
 * array: https://www.avrfreaks.net/forum/putting-variables-eeprom

I attached for you a patch sample using array (simplest way) mapped to eeprom.
[eeprom-stif2.diff.gz](https://github.com/cbalint13/tinnymodbus/files/9960889/eeprom-stif2.diff.gz)
 
Notice the presence of values @0x04000000 (.eeprom) as ```uint8_t  EEMEM EEData[4] = {0x01, 0xfa, 0xce, 0x55};```:
```
$ cat boot.eep 
:04000000 01FACE55 DE
```

> 
> ```
> Disassembly of section .eeprom:
> 
> 00810000 <HumidityOffsetEE>:
> 	...
> 
> 00810001 <IDModbus>:
>   810001:	Address 0x0000000000810001 is out of bounds.

* Seems that overlap is due to mention that EEMEM attribute label maps all vars to the eeprom beginning, hence the overlap.

> ```
> 
> Unfortunately i dont know where to adjust the bonds and position of the eeprom..
> 
> Cheers, Stefan

Hope it helps !

stif, 2022-11-08:
Hello and thanks for your great input!

>Instructions of AVR ISA are 16 bit long, so PC pointer is also 16bit long, CPU always steps one instruction over 16bit (2 octets).

Hm ok, does this mean for some reason the lower 8 bits of the 16bit value are thrown away, and thats why i have to divide by 2?
I am still stuck with the 8bit word to 16bit value transformation on a theoreticical level, but i dont care as long as i know how it is done 🙈.

Also the `0x0000` vs `0x000F` startpoint is not my biggest concern as long as i know one of these versions is working..

But the hint, only a single variable/array/struct with the EEMEM attribute is allowed is great, and solved my problem! It works now - without you i would still searching for a solution.

So thanks again for your help,
cheers,
Stefan

cbalint13, 2022-11-08:
> Hello and thanks for your great input!
> 
> > Instructions of AVR ISA are 16 bit long, so PC pointer is also 16bit long, CPU always steps one instruction over 16bit (2 octets).
> 
> Hm ok, does this mean for some reason the lower 8 bits of the 16bit value are thrown away, and thats why i have to divide by 2? I am still stuck with the 8bit word to 16bit value transformation on a theoreticical level, but i dont care as long as i know how it is done see_no_evil.

* The "footprint" of AVR is 16bit (2 octet wide), for one single CPU "step"  there will be two octets covered & consumed .

> Also the `0x0000` vs `0x000F` startpoint is not my biggest concern as long as i know one of these versions is working..
>
> But the hint, only a single variable/array/struct with the EEMEM attribute is [...]

The definition of EEMEM is: [#define EEMEM __attribute__((section('.eeprom')))](https://electronics.stackexchange.com/questions/172719/define-eemem-attribute-section-eeprom)

 I am also disappointed by this EEMEM behaviour. It instructs compiler (GCC) to pack this variable into a special section (.eeprom) but don't understand why the compiler is not smart enough to allocate non-overlapping slots in this section too handling multiple independent variables that are being tagged here. But that would be another story to learn about some insights of GCC logic :)

> So thanks again for your help, cheers, Stefan

Well, I also learnt new things about some overlooked behaviour, thank you too for revisiting me through these, it is always fun !

~Cristian.